aitjcize/QCamber #6: Support latest ODB++ version

aitjcize, 2015-05-03:
The current ODB++ parser only supports version B.04 of the ODB++ file format. We should make the parser compatible with older version and latest version of the spec.

yodalee, 2015-05-06:
Where can we get the specification?

aitjcize, 2015-05-06:
Yes you need to become a member to download the specification. I thought you are working on the Gerber parser?

yodalee, 2015-05-07:
Yes, just curious

RashidAbbasi, 2019-09-16:
How to read data from ODB++ using C #, any one help me, i dont understand the whole concept. plz 
rashidd.abbasi@gmail.com

koenraad-voltza, 2019-10-22:
These documents seem to be freely available now:
https://www.odb-sa.com/wp-content/uploads/ODB_Format_Description_v7.pdf
https://www.odb-sa.com/wp-content/uploads/odb_spec_user-2.pdf

I'm specifically interested in using v7, can someone give a working dataset to use for experimenting?
aitjcize, I assume you mean v8.0 update 4