aitjcize/QCamber #12: Running error

nana88, 2016-09-30:
Hi,

I build successfully (No error) and when I run it, it's different to the pre-build version. Could you have any idea why?

-My version
https://drive.google.com/file/d/0B2QU5In0ClP0M2Y2bnNkR3U2SVU/view?usp=sharing

-Pre-build version
https://drive.google.com/file/d/0B2QU5In0ClP0MTczLUphN29XNGs/view?usp=sharing

Yours sincerely,
Nana,

aitjcize, 2016-09-30:
Hi what version of Qt are you using?
This was written back in Qt 4.8, can you try it?
Patches are welcome if you want to help port it to 5.x

aitjcize, 2016-09-30:
Also it looks like the color setting is missing in your version. Try settings -> color and set the correct color for each layer.

nana88, 2016-09-30:
Hi, 
I had many errors with 5.x, and I found out that some parts could not run with 5.x so I use Qt 4.8.2 and could build it successfully now.
The color setting is the right answer. 
Thank you very much ^^

P/S: I could not build the files from bison and flex by using Qt. I have no idea why. I built them in cmd of Windows first, and then I build the project in Qt, and it works fine.

Best Regards,
Nana,

aitjcize, 2016-09-30:
Yes you should be using 4.8.2 :)

I'll close this. Please re-open if you have further questions.

closed 2016-09-30 by aitjcize

nana88, 2016-10-04:
Hi,

I am sorry. I have another question, or could I say it is a problem. 
While running it, except 1 layer, all other layers could not read the information and have error like this:

> QFSFileEngine::open: No file name specified
> parse: can't open `' for reading

I guess it is the grammar error, am I right?

Best Regards,
Nana,