aitjcize/QCamber #16: Hollo, Could you tell which version of QT and tools you used to compile it?

kunlunzheng, 2020-08-11:
Very nice to see you great project. We can't compile it and has many error.  Could you help to tell the compile environment?

kunlunzheng, 2020-08-11:
Could you tell which version of QT and QT creator you used to compile it in window?

kunlunzheng, 2020-08-11:
After try some QT creator, we can build the exe. The exe could only view silk layer, drill layer, but can't view track, pad layer. Could you tell the reason?

aitjcize, 2020-08-12:
Can you show a screenshot of the interface? All layer should be parsed and show on the left.

kunlunzheng, 2020-08-12:
Many thanks for your reply!
We imported a ODB++ file generated by PADS. But we could only view silkscreen, drill layer. When we click top or other layer will display QCamber stopped working. 
When we imported a file generated by Altium Designer, but it can't handle it. 
Could you help to indicate how to handle these 2 issue?

kunlunzheng, 2020-08-12:
Dear Sir,  
  Please see the picture of the issue. 
  Please help to try the file of the ODB++ file. IEP_ZF_EMC_CM_Small is from PADS. PCB1 is from Altium Designer. 
   
  Many thanks! 
  Allen Zheng

kunlunzheng, 2020-08-12:
Please help to view the picture and file in baidu disk's link.
https://pan.baidu.com/s/1pW7_J1CzTY6reEDrO6XrXw
getting code:hcbo

aitjcize, 2020-08-13:
Hi, the link show 404 for me, can you re-post?

kunlunzheng, 2020-08-13:
I feel you might can't access Baidu disk. I send the issue files to your email. Could you help to view and check it?

kunlunzheng, 2020-08-13:
I found the email is rejected from aitjcize@gmail.com. Could you have other email address?
I also post the files in the following address. And you could input the 4 getting code 1234 to get it.
https://pan.baidu.com/s/1zkhKZUtJbkPAjKP3GbSQLA
4 Getting Code:1234

aitjcize, 2020-08-13:
still can't access. Can you just post the images in github? I only need to take a look at the sidebar.

kunlunzheng, 2020-08-13:
Hollo, please view the picture and files in the following github link.
https://github.com/kunlunzheng/ODB-issue

myhhub, 2021-05-29:
> Hollo, please view the picture and files in the following github link.
> https://github.com/kunlunzheng/ODB-issue

1.PCB1.tar.gz
   1.1 Bad file format. https://jingyan.baidu.com/article/636f38bb7986cbd6b84610b1.html
   1.2 Missing required attribute name. 'type' 'CONTEXT' not specified.
         file:PCB1\matrix\matrix

         LAYER {
              ROW=14
              NAME=MULTILAYER
              POLARITY=POSITIVE
              START_NAME=
              END_NAME=
              OLD_NAME=
          }
        
2.IEP_ZF_EMC_CM_Small.gz
   2.1 There is a space in the name(Feature symbol name).
       file:IEP_ZF_EMC_CM_Small\steps\pcb\layers\solder-mask-top\features
       $1 oval4300x3200 M

myhhub, 2021-05-29:
[image]
[image]
[image]