aitjcize/QCamber #23: Many compile errors Qt5

dl8dtl, 2025-04-04:
Trying to compile this under an elderly Ubuntu 20.04 (which I unfortunately cannot easily upgrade).
I get many compile errors:
```
cd src/ && ( test -e Makefile || /usr/lib/qt5/bin/qmake -o Makefile /home/WUJ2DR/src/QCamber/src/src.pro ) && make -f Makefile 
make[1]: Entering directory '/home/WUJ2DR/src/QCamber/build/src'
g++ -c -pipe -O2 -Wall -W -D_REENTRANT -fPIC -DQT_NO_DEBUG -DQT_GUI_LIB -DQT_CORE_LIB -I../../src -I. -I../../src -I.build -I../../src/parser -I../../src/parser/odbpp -I../../src/symbol -I../../src/gui -I../../src/graphicsview -isystem /usr/include/x86_64-linux-gnu/qt5 -isystem /usr/include/x86_64-linux-gnu/qt5/QtGui -isystem /usr/include/x86_64-linux-gnu/qt5/QtCore -I.build -I/usr/lib/x86_64-linux-gnu/qt5/mkspecs/linux-g++ -isystem /usr/include/x86_64-linux-gnu/qt5/QtWidgets -o .build/odbppgraphicsminimapview.o ../../src/graphicsview/odbppgraphicsminimapview.cpp
In file included from ../../src/graphicsview/odbppgraphicsminimapview.cpp:25:
../../src/symbol/symbolfactory.h: In static member function 'static Symbol* SymbolFactory::create(const QString&, const Polarity&, const AttribData&)':
../../src/symbol/symbolfactory.h:129:14: warning: catching polymorphic type 'class InvalidSymbolException' by value [-Wcatch-value=]
  129 |     } catch (InvalidSymbolException) {
      |              ^~~~~~~~~~~~~~~~~~~~~~
../../src/graphicsview/odbppgraphicsminimapview.cpp: In constructor 'ODBPPGraphicsMiniMapView::ODBPPGraphicsMiniMapView(QWidget*)':
../../src/graphicsview/odbppgraphicsminimapview.cpp:56:44: error: no matching function for call to 'ODBPPGraphicsMiniMapView::connect(QScrollBar*, const char [19], ODBPPGraphicsMiniMapView*, const char [27])'
   56 |       this, SLOT(updateLayerViewport(void)));
      |                                            ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qsharedpointer_impl.h:64,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qsharedpointer.h:48,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qpixmap.h:48,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qpainter.h:48,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qgraphicsview.h:45,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QGraphicsView:1,
                 from ../../src/graphicsview/odbppgraphicsminimapview.h:26,
                 from ../../src/graphicsview/odbppgraphicsminimapview.cpp:23:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:208:36: note: candidate: 'static QMetaObject::Connection QObject::connect(const QObject*, const char*, const QObject*, const char*, Qt::ConnectionType)'
  208 |     static QMetaObject::Connection connect(const QObject *sender, const char *signal,
      |                                    ^~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:208:59: note:   no known conversion for argument 1 from 'QScrollBar*' to 'const QObject*'
  208 |     static QMetaObject::Connection connect(const QObject *sender, const char *signal,
      |                                            ~~~~~~~~~~~~~~~^~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:211:36: note: candidate: 'static QMetaObject::Connection QObject::connect(const QObject*, const QMetaMethod&, const QObject*, const QMetaMethod&, Qt::ConnectionType)'
  211 |     static QMetaObject::Connection connect(const QObject *sender, const QMetaMethod &signal,
      |                                    ^~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:211:59: note:   no known conversion for argument 1 from 'QScrollBar*' to 'const QObject*'
  211 |     static QMetaObject::Connection connect(const QObject *sender, const QMetaMethod &signal,
      |                                            ~~~~~~~~~~~~~~~^~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:463:32: note: candidate: 'QMetaObject::Connection QObject::connect(const QObject*, const char*, const char*, Qt::ConnectionType) const'
  463 | inline QMetaObject::Connection QObject::connect(const QObject *asender, const char *asignal,
      |                                ^~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:463:64: note:   no known conversion for argument 1 from 'QScrollBar*' to 'const QObject*'
  463 | inline QMetaObject::Connection QObject::connect(const QObject *asender, const char *asignal,
      |                                                 ~~~~~~~~~~~~~~~^~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:228:43: note: candidate: 'template<class Func1, class Func2> static QMetaObject::Connection QObject::connect(const typename QtPrivate::FunctionPointer<Func>::Object*, Func1, const typename QtPrivate::FunctionPointer<Func2>::Object*, Func2, Qt::ConnectionType)'
  228 |     static inline QMetaObject::Connection connect(const typename QtPrivate::FunctionPointer<Func1>::Object *sender, Func1 signal,
      |                                           ^~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:228:43: note:   template argument deduction/substitution failed:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h: In substitution of 'template<class Func1, class Func2> static QMetaObject::Connection QObject::connect(const typename QtPrivate::FunctionPointer<Func>::Object*, Func1, const typename QtPrivate::FunctionPointer<Func2>::Object*, Func2, Qt::ConnectionType) [with Func1 = const char*; Func2 = const char*]':
../../src/graphicsview/odbppgraphicsminimapview.cpp:56:44:   required from here
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:228:43: error: no type named 'Object' in 'struct QtPrivate::FunctionPointer<const char*>'
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:260:13: note: candidate: 'template<class Func1, class Func2> static typename std::enable_if<((int)(QtPrivate::FunctionPointer<Func2>::ArgumentCount) >= 0), QMetaObject::Connection>::type QObject::connect(const typename QtPrivate::FunctionPointer<Func>::Object*, Func1, Func2)'
  260 |             connect(const typename QtPrivate::FunctionPointer<Func1>::Object *sender, Func1 signal, Func2 slot)
      |             ^~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:260:13: note:   template argument deduction/substitution failed:
../../src/graphicsview/odbppgraphicsminimapview.cpp:56:44: note:   candidate expects 3 arguments, 4 provided
   56 |       this, SLOT(updateLayerViewport(void)));
      |                                            ^
In file included from /usr/include/x86_64-linux-gnu/qt5/QtCore/qsharedpointer_impl.h:64,
                 from /usr/include/x86_64-linux-gnu/qt5/QtCore/qsharedpointer.h:48,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qpixmap.h:48,
                 from /usr/include/x86_64-linux-gnu/qt5/QtGui/qpainter.h:48,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/qgraphicsview.h:45,
                 from /usr/include/x86_64-linux-gnu/qt5/QtWidgets/QGraphicsView:1,
                 from ../../src/graphicsview/odbppgraphicsminimapview.h:26,
                 from ../../src/graphicsview/odbppgraphicsminimapview.cpp:23:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:269:13: note: candidate: 'template<class Func1, class Func2> static typename std::enable_if<(((int)(QtPrivate::FunctionPointer<Func2>::ArgumentCount) >= 0) && (! QtPrivate::FunctionPointer<Func2>::IsPointerToMemberFunction)), QMetaObject::Connection>::type QObject::connect(const typename QtPrivate::FunctionPointer<Func>::Object*, Func1, const QObject*, Func2, Qt::ConnectionType)'
  269 |             connect(const typename QtPrivate::FunctionPointer<Func1>::Object *sender, Func1 signal, const QObject *context, Func2 slot,
      |             ^~~~~~~
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h:269:13: note:   template argument deduction/substitution failed:
/usr/include/x86_64-linux-gnu/qt5/QtCore/qobject.h: In substitution of 'template<class Func1, class Func2> static typename std::enable_if<(((int)(QtPrivate::FunctionPointer<Func2>::ArgumentCount) >= 0) && (! QtPrivate::FunctionPointer<Func2>::IsPointerToMemberFunction)), QMetaObject::Connection>::type QObject::connect(const typename QtPrivate::FunctionPointer<Func>::Object*, Func1, const QObject*, Func2, Qt::ConnectionType) [with Func1 = const char*; Func2 = const char*]':
../../src/graphicsview/odbppgraphicsminimapview.cpp:56:44:   required from here
[... cut ...]
   34 |   QTextCodec::setCodecForCStrings(QTextCodec::codecForName("UTF-8"));
      |               ^~~~~~~~~~~~~~~~~~~
make[1]: *** [Makefile:1890: .build/main.o] Error 1
make[1]: Target 'first' not remade because of errors.
make[1]: Leaving directory '/home/WUJ2DR/src/QCamber/build/src'
make: *** [Makefile:47: sub-src-make_first] Error 2
make: Target 'first' not remade because of errors.
```
Project was configured and built with
```
mkdir build
cd build
qmake -makefile ../qcamber.pro
make -k
```

aitjcize, 2025-04-11:
This app was build onto of Qt 4.7. I've never tried compiling for Qt 5. You are welcome to help port it to Qt 5.0.

dl8dtl, 2025-04-11:
Oops. Qt5 is EOL'ed …

dl8dtl, 2025-04-13:
OK, I did it, #24

Qt6 is still a bit away though. However, right now, I have issues at runtime (files selected are then not found) which I'd like to resolve first.

dl8dtl, 2025-04-14:
Even better: contribution by Bernd Brand: #27